RaspAP/raspap-webgui #1749: Dashboard redesign with Bootstrap v5.3

billz, 2025-02-06:
## Is your feature request related to a problem? 
RaspAP's existing dashboard, while clean and functional, has been essentially unchanged for many years. It could be improved by displaying relevant information more intuitively while also being more visually appealing. 

## Describe the solution you'd like
Mockups have been developed that overhaul the dashboard to give it an "at-a-glance" summary of the router's current configuration. Active routes and services are indicated in color; inactive ones are greyed-out. The inactive routes used a dashed rather than solid line. 

Specific requirements are summarized below:

1. A minimum, three (3) different Bootstrap 5.x media sizes should be implemented. These are provided in the following mockups: 1) full width desktop, 2) medium width, and 3) mobile.
2. The device in .svg format using the default theme color is [available here](https://github.com/RaspAP/raspap-webgui/pull/1748/files#diff-998031c156cfebea8de2aebb2d533716e4d43ebd3b4da6e96d317190c48f877f).
3. Icons are limited to Font Awesome Free v6.6. It should be obvious from the mockups which ones are used; if clarity is needed just reply in the issue.
4. Font Awesome [stacking](https://docs.fontawesome.com/web/style/stack) should be used to indicate disabled / unavailable items (the firewall in the mockup, for example).
5. CSS should be placed in `app/css/all.css`.
6. The design elements should be contained inside the existing `<div class="card-body">` in `templates/dashboard.php`.
7. The existing `card-body` contents may be removed entirely.
8. Final markup should be validated with a 508 checker.

## Describe alternatives you've considered
N/A.

## Additional context

1. Full width desktop

[image]
2. Medium width

[image]
3. Mobile devices

[image]

billz, 2025-02-06:
/bounty 300

hunxjunedo, 2025-02-06:
/attempt #1748 
@billz looks like the links to the mockups are broken

billz, 2025-02-06:
@hunxjunedo updated, thx

PR #1750 (2025-02-07): feat: Dashboard redesign 
This PR redesigns the dashboard from the mockups

P.S I need some help with the dynamic variables to replace hardcoded client count and fetch device status, I'm not really familiar with the inner workings of this project. Maybe you can point to documentation?

[image]

[image]

[image]

/claim #1749
/closes #1749
closed 2025-03-15 by billz