RaspAP/raspap-webgui #1762: [Bug]: New versions of 'iw phy#0 info' provides frequencies output with decimals

juliovegap, 2025-02-16:
### Code of Conduct

- [x] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [x] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [x] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [x] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [x] I have searched this repository for existing issues.
- [x] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [x] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [x] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Debian	Bookworm

### Quick install or Manual setup?

Manual setup

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

Other

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

juliovega@gmail.com

### Bug description

Current parser located at `src/RaspAP/Parsers/IwParser.php` cannot process an `iw phy#0 info` output with decimals, for example, the chain `5240.0 MHz [48] (20.0 dBm)` is currently filtered on RaspAP v3.2.7

My proposed solution is to add a dot on the first \d digit matches. I have checked it worked on my setup an shall be compatible with other iw versions
PRE:  `$pattern = '/\*\s+([\d]+)\s+MHz \[(\d+)\] \(([\d.]+) dBm\)\s(?!' .$excluded_pattern. ')/';`
POST: `$pattern = '/\*\s+([\d.]+)\s+MHz \[(\d+)\] \(([\d.]+) dBm\)\s(?!' .$excluded_pattern. ')/';`

### Steps to reproduce

`> iw phy0 info`
`                Frequencies:`
`                        * 2412.0 MHz [1] (20.0 dBm)`
`                        * 2417.0 MHz [2] (20.0 dBm)`
`                        * 2422.0 MHz [3] (20.0 dBm)`
`                        * 2427.0 MHz [4] (20.0 dBm)`
`                        * 2432.0 MHz [5] (20.0 dBm)`
`                        * 2437.0 MHz [6] (20.0 dBm)`
`                        * 2442.0 MHz [7] (20.0 dBm)`
`                        * 2447.0 MHz [8] (20.0 dBm)`
`                        * 2452.0 MHz [9] (20.0 dBm)`
`                        * 2457.0 MHz [10] (20.0 dBm)`
`                        * 2462.0 MHz [11] (20.0 dBm)`
`                        * 2467.0 MHz [12] (20.0 dBm)`
`                        * 2472.0 MHz [13] (20.0 dBm)`
`                        * 2484.0 MHz [14] (disabled)`

### Relevant log output

```shell

```

billz, 2025-02-16:
I'm unable to reproduce this. Please include your iw version, kernel and locale.

```
$ lsb_release -a
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

$ uname -r
6.6.62+rpt-rpi-v8

$ iw --version
iw version 5.19

$ iw phy0 info
 Frequencies:
	* 2412 MHz [1] (20.0 dBm)
	* 2417 MHz [2] (20.0 dBm)
	* 2422 MHz [3] (20.0 dBm)
	* 2427 MHz [4] (20.0 dBm)
	* 2432 MHz [5] (20.0 dBm)
	* 2437 MHz [6] (20.0 dBm)
	* 2442 MHz [7] (20.0 dBm)
	* 2447 MHz [8] (20.0 dBm)
	* 2452 MHz [9] (20.0 dBm)
	* 2457 MHz [10] (20.0 dBm)
	* 2462 MHz [11] (20.0 dBm)
	* 2467 MHz [12] (20.0 dBm)
	* 2472 MHz [13] (20.0 dBm)
	* 2484 MHz [14] (disabled)
````

juliovegap, 2025-02-16:
```
$ lsb_release -a
Description:    Kali GNU/Linux Rolling
Release:        2024.4
Codename:       kali-rolling

$ uname -r
6.6.74-v7l+

$ iw --version
iw version 6.9

$ iw phy0 info
                Frequencies:
                        * 2412.0 MHz [1] (20.0 dBm)
                        * 2417.0 MHz [2] (20.0 dBm)
                        * 2422.0 MHz [3] (20.0 dBm)
                        * 2427.0 MHz [4] (20.0 dBm)
                        * 2432.0 MHz [5] (20.0 dBm)
                        * 2437.0 MHz [6] (20.0 dBm)
                        * 2442.0 MHz [7] (20.0 dBm)
                        * 2447.0 MHz [8] (20.0 dBm)
                        * 2452.0 MHz [9] (20.0 dBm)
                        * 2457.0 MHz [10] (20.0 dBm)
                        * 2462.0 MHz [11] (20.0 dBm)
                        * 2467.0 MHz [12] (20.0 dBm)
                        * 2472.0 MHz [13] (20.0 dBm)
                        * 2484.0 MHz [14] (disabled)

```

billz, 2025-02-17:
Kali Linux is unsupported. However, if your fix is compatible with both versions of iw I'd encourage you to open an PR. This way you'll receive credit in the next release.

billz, 2025-02-19:
Resolved with #1764

closed 2025-02-19 by billz