RaspAP/raspap-webgui #1768: [Bug]: No routing/ip tables on fresh install 3.2.9(img release)

brazier, 2025-02-23:
### Code of Conduct

- [x] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [x] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [x] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [x] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [x] I have searched this repository for existing issues.
- [x] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [x] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [x] I have generated a [RaspAP debug log](https://docs.raspap.com/troubleshooting/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/troubleshooting/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Other

### RaspAP version

3.2.5 (Latest)

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

alexander@linux.com

### Bug description

There is no Routing between eth0 and wlan0, adding iptables roules as per: [manual install](https://docs.raspap.com/manual/#routing-and-ip-masquerading) fixes the issue until next reboot. 

Testet with clean 2.3.9(multiple times) and 2.3.7 install using the supplied images: https://github.com/RaspAP/raspap-webgui/releases/
Running on a [Compute model 3](https://www.compulab.com/products/iot-gateways/iot-gate-rpi-industrial-raspberry-pi-iot-gateway/) Using Raspberry Pi Imager v1.8.5 to flash with setting username, password, timezone, locale and ssh-key.

Doing a fresh install of the official Raspberry OS Lite 64bit(2024-11-19) from the raspberry pi imager with same settings as above, then running the quick installer works without any problems.
` sudo apt-get update
 sudo apt-get full-upgrade
 sudo reboot
 sudo raspi-config
 curl -sL https://install.raspap.com | bash`

### Steps to reproduce

1. [Download .img.zip ](https://github.com/RaspAP/raspap-webgui/releases/)
2. Write using Raspberry PI Imager v1.8.5
3. Connect to AP over wifi (fyi im not using a external adapter)
4. Eth0 connected to other lan(10.0.0.0/24)
5. No routing

[raspap_debug_img.log](https://github.com/user-attachments/files/18931783/raspap_debug_img.log)

[raspap_debug_quick_install.log](https://github.com/user-attachments/files/18931784/raspap_debug_quick_install.log)

### Relevant log output

```shell

```

brazier, 2025-02-23:
_(Also the version should be updated in the form when posting the issue, and perhaps also add the img realese as a install method)_

billz, 2025-02-23:
Inspection of the most recent image [build log](https://github.com/RaspAP/raspap-webgui/actions/runs/13461796605/job/37618685484) revealed the following:

```
Checking iptables rules
/usr/bin/grep: /etc/iptables/rules.v4: No such file or directory
Adding rule: -t nat -A POSTROUTING -j MASQUERADE
iptables: Failed to initialize nft: Protocol not supported
[ x error ]  Unable to execute iptables 
/usr/bin/grep: /etc/iptables/rules.v4: No such file or directory
Adding rule: -t nat -A POSTROUTING -s 192.168.50.0/24 ! -d 192.168.50.0/24 -j MASQUERADE
iptables: Failed to initialize nft: Protocol not supported
[ x error ]  Unable to execute iptables 
Persisting IP tables rules
iptables-save/1.8.9 (nf_tables) Failed to initialize nft: Protocol not supported
[ x error ]  Unable to execute iptables-save 
```

My initial hunch is that the `pi-gen` build needs the following added to `stage2/04-networking/00-packages`:

```
iptables
iptables-legacy
netfilter-persistent
iptables-persistent
```

I'll apply these changes to my fork and test a new build. Thanks for reporting

billz, 2025-02-28:
pi-gen operates in a `chroot` environment on a mounted filesystem instead of an active OS. For this reason, there are no kernel modules loaded. Configuring iptables or nftables fails because there is no real kernel or network stack running.

The solution (obvious now) is to modify our pi-gen fork to configure these rules on first boot.

closed 2025-02-28 by billz