RaspAP/raspap-webgui #1769: Ui toggle to turn on/off access to web interface(and/or other services)

brazier, 2025-02-23:
## Is your feature request related to a problem? 
I was quite supprised to find the web ui open to the wan interface by default, and wonder if there is a conscious decision to make it this way or if it just havent gotten to it yet/prioritizing

## Describe the solution you'd like
A toggle button to turn on/off wan access to many used services web ui(80/443) and possibly ssh(22) or block all.

## Describe alternatives you've considered
at least bind lighttpd to 10.3.141.1 

## Additional context
I might have overlooked some issue and i know [it is in the docs](https://docs.raspap.com/faq/#how-do-i-prevent-wan-access-to-raspaps-web-administration) but i feel it is one of the things new people not know about/over look, and might be a security concern. Usercase might also be different, as in already being behind a firewall/nat. I have just also noticed as an insider you have access to an firewall feature. that might be coming in the future.

billz, 2025-03-02:
Good suggestion. I agree this would be a nice addition to the project.

Based on informed surveys done in the past, the vast majority of RaspAP users operate their devices behind an existing router often in a home network type environment. However, it's still a question/request that pops up now and then. The Insiders firewall plugin makes this a non-issue, but we could still provide a basic level of isolation (binding lighttpd to the AP interface, as you suggested). An option to restrict WAN access could live on the System > Advanced tab.

brazier, 2025-03-02:
> Good suggestion. I agree this would be a nice addition to the project.
...
 That's what i thought, my use case is a travel router/Swiss army knife for work and travel and will be plugging into all sorts of networks, for now i have just added iptables drop all on the eth0 interface not even alowing pings.
`Iptables -A INPUT -i eth0 -j DROP`

closed 2026-11-15 by billz