RaspAP/raspap-webgui #1780: [Bug]: Early error, no IP assigned initially. Unable to connect!

naltabra, 2025-03-07:
### Code of Conduct

- [x] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [x] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [x] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [x] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [x] I have searched this repository for existing issues.
- [x] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [x] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [x] I have generated a [RaspAP debug log](https://docs.raspap.com/troubleshooting/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/troubleshooting/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(32-bit) Lite Bookworm

### Installation method

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 3 Model B

### RaspAP version

Latest

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

Naltabra@gmail.com

### Bug description

Hello,

I’ve been working on this for two days, trying to get the SoftAP up and running. I’m not a Linux person, just networking, and I’ve been relying on ChatGPT to figure things out.

I’m running this on a Raspberry Pi 3 and have tested the current 32-bit and 64-bit Lite versions, as well as the 32-bit Desktop version (Bookworm), all with the same results. The only connected devices are a USB keyboard and an Ethernet cable. wlan0 is not configured in any way.

I don’t know if this is an unfortunate combination of Pi 3, the current Pi distribution, and SoftAP. 
I could really use some help…

/Olof

### Steps to reproduce

Steps taken:
	1.	The SD card was created using Raspberry Pi’s official tool, and the only settings configured this way were hostname and country.
	2.	Booting up the Pi 3 with only eth0 and a USB keyboard connected.
	3.	Followed the setup guide exactly to upgrade the system and then rebooted.
	4.	Started the curl installation, which completed successfully. I disabled all VPN options and Adblock during installation.
	5.	Disconnected the Ethernet cable and rebooted.
	6.	The system comes up and is visible as a WiFi network as expected. However, when I try to connect, it just hangs on “Connecting”.

I started investigating the logs and saw that the WiFi handshake works fine. At this point, it should be assigned an IP address, but this is where it fails.
	•	Checked dnsmasq.conf – looks correct. wlan0 is listed, and the correct IP addresses are set for IF/24.
	•	Checked dnsmasq-dhcp.log – the DHCP request is received, but dnsmasq complains that wlan0 has no IP address, so it can’t assign one either.
	•	Checked ifconfig – wlan0 is up, but it has no IP.
	•	Restarted all relevant services after installation, but it didn’t help.

Additional testing:

When running the Desktop version, I tried setting a static IP manually, and that allowed me to connect. However, I then ran into a different issue – I couldn’t reach the internet.

I feel like I’m staring at the issue but don’t understand how it works at a low level or how it’s implemented in SoftAP. I can reproduce the issue every time on this Pi 3 with the same result.

### Relevant log output

```shell

```

billz, 2025-03-07:
Per the docs, this is completely unsupported.

naltabra, 2025-03-07:
What is unsupported?