RaspAP/raspap-webgui #1809: [Bug]: Parse error

TMTYD, 2025-03-24:
### Code of Conduct

- [x] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [x] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [x] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [x] I have followed the [project prerequisites](https://docs.raspap.com/quick_start/#quick-install).
- [x] I have searched this repository for existing issues.
- [x] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [ ] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [x] I have generated a [RaspAP debug log](https://docs.raspap.com/troubleshooting/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/troubleshooting/#diagnosing-problems).

### Operating System

Debian Bookworm

### Installation method

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

Latest

### Other software or services running with RaspAP?

No other software

### Bug description

RaspAP Exception
An exception occurred
Stack trace:

ParseError: syntax error, unexpected '=>' (T_DOUBLE_ARROW)
 at {closure}(Dashboard.php:30)
 at spl_autoload_call(Unknown Source)
 at DisplayDashboard(dashboard.php:14)
 at handleCorePageAction(page_actions.php:26)
 at require_once(page_actions.php:12)
 at (main)(index.php:119)

### Steps to reproduce

after the update

### Relevant log output

```shell

```

TMTYD, 2025-03-24:
`
[2025-03-24 23:16:06] syntax error, unexpected '=>' (T_DOUBLE_ARROW) in /var/www/html/src/RaspAP/UI/Dashboard.php on line 30`

this is line 30

```
class Dashboard {

    private string $firewallConfig;

    public function __construct() {
        $this->firewallConfig = RASPI_CONFIG.'/networking/firewall.conf';
    }

    /*
     * Returns the management page for an associated VPN
     *
     * @param string $interface
     * @return string
     */
    public function getVpnManged(?string $interface = null): ?string
    {
        return match ($interface) {
            'wg0' => '/wg_conf',
            'tun0' => '/openvpn_conf',
            'tailscale0' => '/plugin__Tailscale',
            default => null,
        };
    }
```

closed 2025-03-25 by billz

TMTYD, 2025-03-26:
I did this with the PHP 7.4 fix.

This: https://github.com/RaspAP/raspap-webgui/pull/1810/commits/20fe5fc5a7f54625e4579504e4c54b3f7af9cc0c

Now I keep getting the message "The session has expired."

[image]

billz, 2025-03-27:
See this https://github.com/RaspAP/raspap-webgui/discussions/1788

TMTYD, 2025-03-27:
Hi

Thanks for the info, but that's already defined. I still get the message.

[image]

TMTYD, 2025-03-28:
@billz can you help pls?

TMTYD, 2025-04-06:
I have now reinstalled the entire system and reinstalled raspap, now there is also an error message with the session

billz, 2025-04-06:
Why not simply use a pre-built custom OS with RaspAP?

PR #1810 (2025-03-25): Fix: php7.4 compatibility
Resolves #1809