RaspAP/raspap-webgui #1826: [Bug]: Broken webui after update

MR0000001, 2025-04-17:
### Code of Conduct

- [x] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [x] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [x] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [x] I have followed the [project prerequisites](https://docs.raspap.com/quick_start/#quick-install).
- [x] I have searched this repository for existing issues.
- [x] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [ ] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [x] I have generated a [RaspAP debug log](https://docs.raspap.com/troubleshooting/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/troubleshooting/#diagnosing-problems).

### Operating System

Debian Bookworm

### Installation method

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 5

### RaspAP version

Latest

### Other software or services running with RaspAP?

Yes (specify below)

### Bug description

Updating from 3.1.3 to 3.3.1 break webui, even after using the unistall & install process it stay broken.

source /var/www/html/installers/uninstall.sh 
_remove_raspap

curl -sL https://install.raspap.com | bash

[image]

[image]

### Steps to reproduce

Starting from 3.1.3 webui, update to 3.3.1.

### Relevant log output

```shell

```

billz, 2025-04-17:
This is not a bug. Clear your browser cache

closed 2025-04-17 by billz

MR0000001, 2025-04-17:
While in anonym goes, in normal run even after hard refresh continues.