RaspAP/raspap-webgui #1844: [Bug]: Prebuilt images do not boot on headless Pi Zero 2 W.

ajd4096, 2025-05-03:
### Code of Conduct

- [x] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [x] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [x] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [x] I have followed the [project prerequisites](https://docs.raspap.com/quick_start/#quick-install).
- [x] I have searched this repository for existing issues.
- [x] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [ ] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [x] I have generated a [RaspAP debug log](https://docs.raspap.com/troubleshooting/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/troubleshooting/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(32-bit) Lite Bookworm

### Installation method

Pre-built image

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi Zero 2 W

### RaspAP version

Latest

### Other software or services running with RaspAP?

Yes (specify below)

### Bug description

Using the 3.3.3 prebuilt image on a headless device fails to finish booting, and no AP is created.

I expected the device to be accessible via the AP as described in the "Initial Settings" section.

(Note: it is not possible to capture a debug log if the OS fails to boot.)

### Steps to reproduce

Using either Pi imager 1.9.0 (with no custom settings) or balenaEtcher 2.1.0

Flash raspap-bookworm-armhf-lite-3.3.3.img to a 32GB SD card.

Put the SD card into a headless Pi Zero 2 W, no monitor cable, no OTG cable.

Power up, wait 5 mins.

Attach the monitor cable and you can see the startup is stalled at the "Please enter new username" prompt.

I have confirmed this with multiple SD cards and multiple Pi Zero 2 W boards.

### Relevant log output

```shell

```

billz, 2025-05-03:
Here's your solution:

[image]

Choose this option with Pi imager and try again.

ajd4096, 2025-05-04:
Using that (or custom.toml which also works with balenaEtcher) to set the username does allow the boot to complete, but ...

The AP is active but does not provide any IP addresses.

/var/log/dnsmasq.log:
Apr 29 15:20:22 dnsmasq-dhcp[877]: DHCP packet received on wlan0 which has no address
Apr 29 15:20:26 dnsmasq-dhcp[877]: DHCP packet received on wlan0 which has no address
Apr 29 15:20:33 dnsmasq-dhcp[877]: DHCP packet received on wlan0 which has no address

ip addr:
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether d8:3a:dd:9a:15:46 brd ff:ff:ff:ff:ff:ff

closed 2025-05-04 by billz

billz, 2025-05-04:
Your issue is of a general nature and not directly related to a defect with this project. See https://docs.raspap.com/issues/#issue-policy