RaspAP/raspap-webgui #1864: [Bug]: Can not access Dekstop environment after install (64-bit OS only)

pud469, 2025-05-21:
### Code of Conduct

- [x] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [x] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [x] I have read and followed the [common sense checklist](https://docs.raspap.com/troubleshooting/#dos-and-donts).
- [x] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [x] I have followed the [project prerequisites](https://docs.raspap.com/quick_start/#quick-install).
- [x] I have searched this repository for existing issues.
- [x] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [ ] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [x] I have generated a [RaspAP debug log](https://docs.raspap.com/troubleshooting/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/troubleshooting/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Desktop Bookwom

### Installation method

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi Zero W

### RaspAP version

Latest

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

gooki@frontier.com

### Bug description

Clean easy install on a PI zero W2 64-bit bookworm , prevents LightDM.service from starting.  I have tried this multiple times.  It also has the same outcome on the on the 64 bit legacy (Bullseye) os.  This does not happen on any of the PI OS 32-bit.   I did make a copy of the system logs.  I need to grab another SD card as I do not want to trash my current install (32-bit, working fine with RaspAP)  and I will get the debug information.  I'm sorry I didn't grab that sooner as I wanted to finish my project before I read the info to submit a bug report.  

[RP_64_log.zip](https://github.com/user-attachments/files/20358615/RP_64_log.zip)

### Steps to reproduce

Install Raspberry Pi OS 64-bit on Pi zero W2, and run the Quick installer.  Reboot and be sure to have the HDMI output connected.  The window manager will not run.

### Relevant log output

```shell

```

billz, 2025-05-21:
If I understand this correctly, you didn't observe this issue on 32-bit Desktop (bookworm), correct?

pud469, 2025-05-23:
That is correct.  Only on Pi Zero W2 64-bit.  I should have another SD card tomorrow so I can reinstall and get you the debug info.  I also want to get more error details from the system.  I'm sorry from the lack of info, I just wanted to finish what I was doing, and if I didn't send the bug report, I would have forgotten about it.

billz, 2025-05-23:
No problem, thanks for clarifying. I don't have time to diagnose or fix this, so the expedient solution is to drop support for this distro.

pud469, 2025-05-23:
Sure no problem, I wish I was smarter to help quickly.  At least others will know to expect this.  I'll dig through what I can and see if I can at least fix the issue.  If you can at least provide a way to still install this with the easy method, so I can continue to troubleshoot.  I'm sure I'll break things where I will need to reinstall.  If you can possibly add a system check and answer dialog that its broken, instead of removing it entirely?  Or maybe just leave it since I'm the only one to make a bug report?? I'm a controls engineer and love learning new code.  But it will be slow for me.  If I can figure something, even if it is a post install script I'll let you know.  I want to help if I can, this is just a totally new environment for me.  Including on how to properly use github.  I have built shell scripts (typically bash) for others at work using linux.  I'm on staycation for the next week, and this makes for some good wind down time after working around the house.  My first approach will be to get lightdm to work after the install, but then I'll go back to the code and find out which part is causing the issue.  From reading it looks like you are the only one on this project.  If I do come up with some kind of fix, I'm sure I won't submit it correctly lol.  I have attached the debug log, and the most I know how to get better details from the system, just in case there is something there you would like to see.  Thank you for the responses, and thank you for the awesome work!

[sys_journ.log](https://github.com/user-attachments/files/20415961/sys_journ.log)
[raspap_debug.log](https://github.com/user-attachments/files/20415962/raspap_debug.log)

billz, 2025-05-24:
The Zero 2 W has only 512MB RAM which is critically low for running a full desktop environment with LightDM. Is there a reason you're using this hardware? Did you try running the desktop environment without RaspAP?

The rapid restart attempts in `sys_journ.log` suggest a fundamental configuration issue rather than a temporary resource problem. Most likely causes are missing display drivers or insufficient memory allocation for the GPU. RaspAP is very lightweight in terms of resource usage, so I'd be surprised if this exhausted what little RAM is probably left on your Zero.

pud469, 2025-05-25:
It is to use Remmina on the Pi but also create an AP for other users.  I do not really need the DE, and/or use the 32-bit version.  It is also a hobby that I can use at work.  During our cutovers we can benefit from wireless functionality, and for myself to also have RDP.  Painfully slow is quite OK, and don't mind burning the SDcard with a swap file.  Tinkering is fun.

Anyway, I have installed RaspAP from the desktop environment and also over SSH as part of troubleshooting. Probably re-imaged the 64-bit version sdcard 4-5 times to confirm this before reporting.  What  I have just found that  **after** the install of RaspAP on the 64 bit version, /usr/share/xgreeters/pi-greeter-labwc.desktop is missing.  (also pi-greeter-wayfire.desktop).  I even grabbed the PI OS images (.img.xz files) and verified they come with these included, along with a clean 64-bit install on the zero w2, but after using RaspAP easy installer, they are missing.  Something is happening that these are getting removed.  This bug possibly only occurs using wayland (now default), vs xorg, as pi-greeter.desktop (xorg) remains after the install.  It seems more likely there is a dependency at fault here when installing RaspAP.  I'm going to try X tomorrow, but still curious enough to keep looking even if X does work, I want to try and figure out why those files go away. During my original searching I found a few other posts with the same error message, with no mention of RaspAP, so I wonder if my digging could help a few others also.

billz, 2025-05-25:
Never heard of Remmina. I'm confused... if you don't need the DE why are you struggling with it? And why are you running desktop on a Pi Zero?

If you suspect RaspAP's dependencies are somehow at fault, you can try installing the packages manually:
https://github.com/RaspAP/raspap-webgui/blob/98f11477b0279bf7e8a8d7228c6ab57423f5e4e4/installers/common.sh#L224

Tinkering is fine, I get it. But without reproducible evidence my hunch is there's nothing to fix here.

closed 2025-05-29 by billz

billz, 2025-06-01:
Try reinstalling your desktop interface with `sudo apt reinstall raspberrypi-ui-mods lxsession` (untested)

pud469, 2025-06-04:
Sorry I got caught up with some other stuff.  The pi zero w2 is an OK experience running a DE.  Where I would be using this device at work, not everyone is comfortable with a terminal.  The primary purpose is to be to run RaspAP, but there are a few other things I would like to install once I can get through this and the DE would be useful.  Remmina is just and RDP/VNC/SSH client, but I never tried using it for SSH.  I filled the bug report since I would like you to know, and try to help.  I started installing manually step by step, and just recently finished installing the dependencies 1 at a time, restarting after every one to see if something there was a cause.  So far everything is fine, so there goes that theory.  I would be willing to record the installation, just to show what I encountered. That is, if your are interested in seeing it.  If you can not replicated yourself, whether it be because of time, or it just doesn't happen, that is OK, and not expecting or even asking for a fix.  I'm probably one of the very few running a DE on a pi zero, but its fun for me.  Also, it would be nice if I can contribute to the 4 others running a 64-bit install on a pi zero w2 with a DE lol.  I really appreciate your time and communication and I will be sure to update you if I ever find anything. Thank you!!