RaspAP/raspap-webgui #1879: [Bug]: Lagging & Crash on Custom Blocklist

Graham1808, 2025-06-17:
### Code of Conduct

- [x] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [x] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [x] I have read and followed the [common sense checklist](https://docs.raspap.com/troubleshooting/#dos-and-donts).
- [x] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [x] I have followed the [project prerequisites](https://docs.raspap.com/quick_start/#quick-install).
- [x] I have searched this repository for existing issues.
- [x] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [x] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [x] I have generated a [RaspAP debug log](https://docs.raspap.com/troubleshooting/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/troubleshooting/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Desktop Bookwom

### Installation method

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

Latest

### Other software or services running with RaspAP?

Yes (specify below)

### Bug description

Hi When trying to copy Hagezis ultimate blocklist the page hangs for ages after clicking save it then loads a page saying Invalid CSFW token. IT then doesn't save any of the data put in. 
The only way i could get it to save the data way by Sudo nano direct into the config file. After saving the page hanges for ages whilst loading and still shoots out the CSFW error when trying to make changes.
Whole adblock tab is very very slow to load.

### Steps to reproduce

Copy custom list into adblocking and click save

### Screenshots

[image]

### Relevant log output

```shell

```

Graham1808, 2025-06-17:
> > Hagezis ultimate blocklist
> 
> RaspAP supports [these blocklists](https://github.com/RaspAP/raspap-webgui/blob/master/config/blocklists.json). Hagezis is not one of them. Did you manually update the blocklist config?

Yes manually copied the list from github into the custom blocklist config file using nano as when trying to complete via the GUI I get the csfw error and nothing saves.
Not sure if it's due to the size of the list but the whole adblock tab now is really unresponsive

billz, 2025-06-17:
@Graham1808 this has been added as a future enhancement, thanks

closed 2025-06-17 by billz