RaspAP/raspap-webgui #1892: Auto-set country code

Kzitold, 2025-07-06:
You can get the selected country code with `sudo raspi-config nonint get_wifi_country`

Maybe copy that output to the RaspAP country code?

billz, 2025-07-06:
I misread your feature request. Yes, the country code could be set this way.

billz, 2025-07-07:
`raspi-config` returns the regulatory domain value that gets applied at boot time. you can find this in `/proc/cmdline`:

```
cfg80211.ieee80211_regdom=US
```

In my case, I set to "US" in the RPi Imager tool. RaspAP uses `iw reg get/set` (as above) which interacts with different parts of the wireless stack. This is what WLAN drivers and the kernel actually use to determine channel restrictions, etc.

So `hostapd.conf` could initially be populated from `raspi-config`, but thereafter this is ignored and `iw` is used.

Kzitold, 2025-07-07:
As long as hostapd gets on the same country code as Raspberry itself.

closed 2026-11-15 by billz