RaspAP/raspap-webgui #1912: [Feature request] Navbar stays expanded when changing pages on Mobile

ruralmeltdown, 2025-07-23:
Not sure if this is a bug or feature, but when I browse the RaspAP WebUI on mobile, the navigation bar stays expanded when I switch between pages, and I have to close it when the target page loads. It seems it should close after switching pages, as it's a pop-over bar, as opposed to a side-menu that is part of the page, and it obfuscates the page and being able to do anything on it.

billz, 2025-07-23:
This is probably the default behavior of [SB Admin 2](https://github.com/StartBootstrap/startbootstrap-sb-admin-2) which RaspAP implements. Maybe try their demo and see if you observe the same?

ruralmeltdown, 2025-07-23:
It's not the overlaying that's the "problem" (more of an minor annoyance than problem), it's that it persists when you change pages, and have to click the "hamburger" icon to collapse it when the new page loads.

billz, 2025-07-23:
I see what you mean. There's quite a bit going on elsewhere in the project at the moment. If you're comfortable with Bootstrap / frontend dev perhaps open a PR?

ruralmeltdown, 2025-07-23:
I'll see what I can do

ruralmeltdown, 2025-07-23:
Looks like it's just commenting out these lines (tested locally, works as expected, just requires clearing browser/tab cache). I can create a PR if you still want? https://github.com/RaspAP/raspap-webgui/blob/2f9a2dfa92e2dae84bb2eb2c7155dfaeefcd1d5f/dist/sb-admin/js/scripts.js#L15-L18

billz, 2025-07-23:
Good catch. Feel free to open a PR that comments this out. You can also copy code from [this PR](https://github.com/RaspAP/raspap-webgui/pull/1897) to prevent caching, if you like (completely optional).

ruralmeltdown, 2025-07-23:
Hopefully I did that ^ right?

billz, 2025-07-23:
Perfectly, yes :)

billz, 2025-07-23:
Resolved in #1914 @ruralmeltdown many thanks

closed 2025-07-23 by billz