RaspAP/raspap-webgui #1925: [Bug]:

RobotHanzo, 2025-08-12:
### Code of Conduct

- [x] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [x] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [x] I have read and followed the [common sense checklist](https://docs.raspap.com/troubleshooting/#dos-and-donts).
- [x] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [x] I have followed the [project prerequisites](https://docs.raspap.com/quick_start/#quick-install).
- [x] I have searched this repository for existing issues.
- [x] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [ ] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [x] I have generated a [RaspAP debug log](https://docs.raspap.com/troubleshooting/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/troubleshooting/#diagnosing-problems).

### Operating System

Debian Bookworm

### Installation method

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Orange Pi family

### RaspAP version

Latest

### Other software or services running with RaspAP?

Yes (specify below)

### Bug description

After updating to v3.3.9 from v3.3.8, data usage graphs & speedtests are broken with the following errors from devtools console:

`wlan0_info:397  GET http://10.3.141.1/app/js/vendor/dashboardchart.js net::ERR_ABORTED 404 (Not Found)`
`network_conf:473  GET http://10.3.141.1/app/js/vendor/speedtestUI.js net::ERR_ABORTED 404 (Not Found)`

### Steps to reproduce

- Update to 3.3.9
- Access the aforementioned pages
- Broken

### Additional context

With some digging, I believe #1905 broke these features by moving the vendor assets under the `vendor` folder without actually including the `vendor` folder. (Or perhaps this was not installed during the update process?)

### Relevant log output

```shell

```

billz, 2025-08-12:
A fix is already in progress RaspAP/raspap-webgui/pull/1926 
Thanks for reporting

NathanLee402, 2025-08-12:
> In the future, please use a descriptive title for your report

Sorry, I overlooked the title field

billz, 2025-08-12:
Resolved and a new bugfix release version is available. @RobotHanzo thanks!