RaspAP/raspap-webgui #1934: [Bug]: Custom install path not working

gobtrix, 2025-08-19:
### Code of Conduct

- [x] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [x] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [x] I have read and followed the [common sense checklist](https://docs.raspap.com/troubleshooting/#dos-and-donts).
- [x] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [x] I have followed the [project prerequisites](https://docs.raspap.com/quick_start/#quick-install).
- [x] I have searched this repository for existing issues.
- [x] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [ ] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [x] I have generated a [RaspAP debug log](https://docs.raspap.com/troubleshooting/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/troubleshooting/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Installation method

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

Latest

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

freakbsd@gmail.com

### Bug description

Hi team,

Fresh install RaspberryPI OS, fresh install RaspAP on custom path /var/www/html/blablaandbla
After reboot, the following errors (and it's not just that, the paths need to be adjusted in several places.):

2025-08-19 18:03:54: (response.c.203) -- : No such file or directory
2025-08-19 18:03:54: (response.c.206) Path         : /var/www/html/login
2025-08-19 18:03:54: (response.c.208) URI          : /login

# OS info
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

### Steps to reproduce

Fresh install Raspberry Pi and RaspAP. After reboot, mgmt page is not accessible.

### Screenshots

gobtrix, 2025-08-19:
Update:

billz, 2025-08-20:
> If I add the directory to the url, as follows, it shows me the login interface.
> http://10.4.0.2/blablaandbla/login?action=%2Fblablaandbla%2F

This is the way lighttpd (and most web servers) work. anything under `/var/www/html/` is by default part of the document root. Therefore, creating a subdirectory `/var/www/html/mypath` is effectively a subsite and must be accessed as `http://rpitest.local/mypath`

Thanks for reporting.

closed 2025-08-20 by billz